paul1956/CareLink #3: Exception during conversion data

tloczekt, 2022-07-04:
I have a problem with language format of numbers....
" Conversion from string " 213.275" to type "Double" is not valid.
This is error for real data and test sample data as well. I think it comes from my own national local (polish) decimal pointer settings. In my case comma is a decimal pointer not dot. 

More details on attached file
[exception.txt](https://github.com/paul1956/CareLink/files/9038623/exception.txt)
.

paul1956, 2022-07-08:
@tloczekt This should be fixed in version 3.3 I uploaded today. Please report if it is not.

paul1956, 2022-07-08:
@tloczekt it turns out the source data is always English, US and now I should override all the conversions. The output should be local.

tloczekt, 2022-07-11:
Hi, 
@paul1956 thank you so much for your soon response and new release. It's better, I can see data on display, but curve of BG on HOME Page is missing, so BG data are still bad in conversion:  System.FormatException: Input string was not in a correct format. Its interesting because on Runing ActiveInsulin tab curve of insulin and BGs  are visible. 
Other data like present BG, TIR, sensor, vial %, Carbs are OK.
Information of bolus and meals are also ok.
 
[exception2.txt](https://github.com/paul1956/CareLink/files/9083595/exception2.txt)

paul1956, 2022-07-12:
I think all those issues are fixed in 3.4.0.0 but I don’t have a stable internet connection to upload it.  I will try again Thursday if I can’t get it uploaded today otherwise it will be Monday.

 

If you are not in 100% scaling I am having issues painting some of the data the newest release should be better.

Importance: High

tloczekt, 2022-07-13:
3.4.0.0 is step back for me, because again I have none data from the cloud, because String to datetime conversion error.
(System.FormatException: String '13.07.2022 00:01:00' was not recognized as a valid DateTime.)
More details in attached file.
[exception3.txt](https://github.com/paul1956/CareLink/files/9099157/exception3.txt)

paul1956, 2022-07-14:
@tloczekt @BenoitVDBr Thanks for the report, I believe the current version will handle this better in version 3.4.0.2 x64. It turns out that a few dates in the files are localized while most are in US-eng culture. I now try both formats when converting dates and I should also display in local culture with a few exceptions where the data is actually a string that just looks like a date. I am going to leave this open until you both verify it is fixed the two issues are actually different. Also, some of the data being truncated had to do with internally using a comma as a field separator and when numbers have a comma it breaks the algorithm.

BenoitVDBr, 2022-07-14:
Hello Paul,

Thanks for the quick response. I tried the newest version, but the same problem persists. I do however see that the last SG value (upper left of the window) is retrieved correctly from the Carelink server. In the Active Insulin tab, the correct amount is displayed. Due to the error I don't see anything in the graph screens and I get "???" instead of values under the different icons. The TIR values aren't retrieved correctly.

[image]

I don't know if the new description brings any new info...

[Unhandled Exception 2.txt](https://github.com/paul1956/CareLink/files/9114848/Unhandled.Exception.2.txt)

Wkr,
Benoit.

tloczekt, 2022-07-14:
Hello Paul,
version 3.4.03 x64 seems OK. We have all data and software started without any error message.
Main graph is visible as well. Thank you very much for your soon response and hard work. 
[image]

BenoitVDBr, 2022-07-14:
It works, it works, it works!!

Thank you very much!

closed 2022-07-16 by paul1956