paul1956/CareLink #5: Visual Basic error with europan date format

PhilMEFL, 2022-07-24:
When I try to use Carelink, I immediately receive an error saying: System.FormatException: String '23/07/2022 11:55:18' in ToSgList line 17 was not recognized as a valid DateTime.
When I click OK, the application vanishes.

paul1956, 2022-07-24:
@PhilMEFL which version? Can you drop in the CareLink*.json File that causes the issue here? It is a text file you can view, and all person information is stripped out before being saved. It is in your documents folder.
Also, what is the language/culture code your computer is running in. something like "en_US".

My next release 7/28 will package all of the information for you to make it easier to address language issues.

PhilMEFL, 2022-07-24:
I use the latest version available that I downloaded this morning.  My culture is en-us but the language I use is en-be.  Heere are the json files.
[carelinjson.zip](https://github.com/paul1956/CareLink/files/9175816/carelinjson.zip)
Thanks.

PhilMEFL, 2022-07-24:
This can probably help you;
[image]

paul1956, 2022-07-24:
@PhilMEFL That is the issue, most data in the file is in en-US but a few pieces are in what your pump things is local culture in your case en-be and the windows program can't figure that out. I can't decide if I should use a drop down where people can pick their own or just try all the variations.

I am open to suguestions.

paul1956, 2022-07-25:
@PhilMEFL I have put a text version here. It should have fixed issue and if not it produces an error report with dump as a text file in your documents folder that gives me everything to reproject the issue include all local culture information. All personal information is removed.  You can then just do it here. This is a preview of what I plan to release Friday.
[CareLinkv3.4.2.0.zip](https://github.com/paul1956/CareLink/files/9177326/CareLinkv3.4.2.0.zip)

PhilMEFL, 2022-07-25:
Hi Paul, thank you for trying to fix the issue but I have the exact same problem with the new version.

paul1956, 2022-07-25:
Does it produce a crash report? It would be stored under myDocuments. Name is CareLinkErrorReport.txt

PhilMEFL, 2022-07-25:
No sorry, there is no crash report.

PhilMEFL, 2022-07-25:
This is the only recent file in my documents folder.
[CareLinkLastDownload(en-US).zip](https://github.com/paul1956/CareLink/files/9180025/CareLinkLastDownload.en-US.zip)

paul1956, 2022-07-25:
@PhilMEFL you are correct in the case of this one error program exited without saving error report. I have fixed that
[CareLinkv3.4.2.1x64.zip](https://github.com/paul1956/CareLink/files/9183466/CareLinkv3.4.2.1x64.zip).

PhilMEFL, 2022-07-26:
Here is the crash report file
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/9187229/CareLinkErrorReport.en-US.txt)
Thank you for your efforts..

paul1956, 2022-07-26:
@PhilMEFL thank you that helped tremendously, turns out I still had a few CDate conversion what did not understand Cultures. Your pump puts out dates in your countries culture, but your Windows is in English, so the conversion was not happening, and app crashed.

[CareLinkv3.4.2.2x64.zip](https://github.com/paul1956/CareLink/files/9187818/CareLinkv3.4.2.2x64.zip)

I believe this finally fixes your issue.

PhilMEFL, 2022-07-26:
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/9188179/CareLinkErrorReport.en-US.txt)
Sorry to disappoint you but the problem is still not fixed.  However, I succeeded to run the application by changing my location and default language to Belgian French.
Nevertheless, I hope that you will find a solution because although French-speaking, I have trouble communicating in French with my computers, professional deformation no doubt.

paul1956, 2022-07-26:
@PhilMEFL I have 2 solutions load all 110 unique culture data formats which will slow the program down when pump culture is not your Windows UI culture or I could add a dropdown where you select your pump culture. What would you prefer?

paul1956, 2022-07-26:
@PhilMEFL 
This version supports 110 cultures see if it fixes it otherwise, I am going to have to do a dropdown list.

[CareLinkv3.4.2.3x64.zip](https://github.com/paul1956/CareLink/files/9188739/CareLinkv3.4.2.3x64.zip)

PhilMEFL, 2022-07-26:
Sorry but the same problem occurs:
[image]
There is no error log in my documents.

paul1956, 2022-07-26:
@PhilMEFL It turns out every time new data was read one culture was being deleted. So eventually all 110 cultures disappear and that doesn't show up using an error file only on a real system.
[CareLinkv3.4.2.4x64.zip](https://github.com/paul1956/CareLink/files/9193835/CareLinkv3.4.2.4x64.zip)
 
Hopefully this finally fixes the issue. Thanks again for all your help.

PhilMEFL, 2022-07-27:
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/9197023/CareLinkErrorReport.en-US.txt)
Sorry Paul, still the same issue

paul1956, 2022-07-28:
@PhilMEFL  the only way I know to solve your specific issue is to debug in on your system under Visual Studio and some form of remote desktop or if you are a programmer, you can look at it the project is open source. Sorry.

Krikke99, 2022-08-01:
Hi @paul1956 
I was able to reproduce this error, by setting windows language and region to Nederlands (Nederland). I use Nederlands(België) witout errors.
Error report is included:

[image]

[CareLinkErrorReport(en-BE).txt](https://github.com/paul1956/CareLink/files/9232153/CareLinkErrorReport.en-BE.txt)
.

paul1956, 2022-08-03:
@Krikke99 I have an experimental version with a complete rewrite of date handling code. I will post here if to you to try.  It is very early. The version has wrong, and ReadMe not updated. Some dates may display in wrong time zone, but nothing should crash.
[CareLinkv3.4.3.0x64.zip](https://github.com/paul1956/CareLink/files/9247722/CareLinkv3.4.3.0x64.zip)
.

Krikke99, 2022-08-03:
@paul1956 i tested your preview version and no crash occurred.

PhilMEFL, 2022-08-03:
Hi Paul,
This release is working.  Thanks.
Philippe

paul1956, 2022-08-03:
@PhilMEFL Can you check that the dates are correct on your system? It's great that it doesn't crash but are the dates correct? I am still working on showing the date native to your pump and then translated dates relative to the PC. So, if someone in Germany is monitoring someone if US the information is correct and understandable. I will release this version next weekend after more testing.

PhilMEFL, 2022-08-03:
The date is formatted so: 2022-08-03T11:14:00.000Z but I don’t mind such notation.  However, the status of the battery and the sensor remain unknown as you can see in the screenshot.
***@***.***
Thanks

paul1956, 2022-08-03:
Battery issue is fixed is this beta
[CareLinkv3.4.3.0x64.zip](https://github.com/paul1956/CareLink/files/9251357/CareLinkv3.4.3.0x64.zip)

If you are happy with the fixes please close issue.

closed 2022-08-03 by PhilMEFL

PhilMEFL, 2022-10-11:
Hi Paul,
I think the dropdown solution will be the easiest one.  I checked my pump but I don’t find a way to change its culture, only the language can be changed.  I suppose they provide different variant depending on the area where it will be used.

***@***.***

Philippe Martin - 0495 408 883
rue de la Bruyère d’Huleu, 2 - 1460 Ittre
***@***.******@***.***>
IBAN: BE08 7512 0974 0413 - BIC: AXABBE22 - T.V.A. : BE0721.449.079

paul1956, 2022-10-11:
@PhilMEFL I think this issue is fixed, this is not a pump issue it's my application. I now can find the pumps Culture and if there is still an issue, I can use it.