paul1956/CareLink #6: "The given key 'MMOL_L' was not present in the dictionary."

Komle92, 2022-07-25:
First of all, thanks for making this, it seems incredibly practical!

As soon as I log in though, i get this error: "The given key 'MMOL_L' was not present in the dictionary". I'm guessing it's relevant that I'm in Norway, and therefore use mmol/l units, but have to have my Carelink account registered as a US account to avoid the Medtronic mandated 2FA in Norway. After the error appears the app is stuck loading. 

I'm using the newest version (as of 24/7) of CareLink Windows App on Windows 10.

Here is the rest of the exception text:

System.Collections.Generic.KeyNotFoundException: The given key 'MMOL_L' was not present in the dictionary.
   at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at CareLink.UserMessageHandler.GetLocalizedUnits(String unitName) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Utilities\UserMessageHandler.vb:line 81
   at CareLink.Form1.UpdateRegionalData(Dictionary`2 localRecentData) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 1524
   at CareLink.Form1.FinishInitialization() in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 1877
   at CareLink.Form1.DoOptionalLoginAndUpdateData(Boolean UpdateAllTabs) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 1864
   at CareLink.Form1.Form1_Shown(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 133
   at System.Windows.Forms.Form.OnShown(EventArgs e)
   at System.Windows.Forms.Form.CallShownEvent()
   at System.Windows.Forms.Control.InvokeMarshaledCallbackDo(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()

paul1956, 2022-07-26:
It looks like some pumps use MMOLL and other use MMOL_L for mmol/l units,  I now support both in attached beta version

This beta version has a crash handler that will produce a sanitized human readable text file (stored in your documents folder) that will include everything I need to address the issue. It also has much better culture handling for dates there are actually 110 different date formats used around the world. I currently support 12 in this version.
[CareLinkv3.4.2.3x64.zip](https://github.com/paul1956/CareLink/files/9188732/CareLinkv3.4.2.3x64.zip)

Komle92, 2022-08-04:
Thanks for adding support for that. In the new version i don't get stuck anymore so definitely an improvement already. Never considered there would be that many date formats though!

Here is what the text file says (sorry for the long copy-paste, apparently github doesn't support uploading json files):

{
  "lastSensorTS": "0",
  "medicalDeviceTimeAsString": "04/08/2022 17:21:01-00:00",
  "lastSensorTSAsString": "04/08/2022 17:18:01",
  "kind": "BLENGP",
  "version": "2",
  "pumpModelNumber": "MMT-1885",
  "currentServerTime": "1659626556185",
  "lastConduitTime": "0",
  "lastConduitUpdateServerTime": "1659626443227",
  "lastMedicalDeviceDataUpdateServerTime": "1659626443227",
  "firstName": "Alicia",
  "lastName": "Mobile",
  "conduitSerialNumber": "BC376D-457F-40D5-B7F6-0000-0000-0000-0000",
  "conduitBatteryLevel": "31",
  "conduitBatteryStatus": "MEDIUM",
  "conduitInRange": "True",
  "conduitMedicalDeviceInRange": "True",
  "conduitSensorInRange": "True",
  "medicalDeviceFamily": "NGP",
  "sensorState": "NO_ERROR_MESSAGE",
  "medicalDeviceSerialNumber": "NG2627338H",
  "medicalDeviceTime": "0",
  "sMedicalDeviceTime": "04/08/2022 17:21:01",
  "reservoirLevelPercent": "10",
  "reservoirAmount": "300",
  "reservoirRemainingUnits": "31.1",
  "medicalDeviceBatteryLevelPercent": "50",
  "sensorDurationHours": "55",
  "timeToNextCalibHours": "5",
  "calibStatus": "LESS_THAN_SIX_HRS",
  "bgUnits": "MMOL_L",
  "timeFormat": "HR_24",
  "lastSensorTime": "0",
  "sLastSensorTime": "04/08/2022 17:18:01",
  "medicalDeviceSuspended": "False",
  "lastSGTrend": "NONE",
  "lastSG": "{\u0022sg\u0022:184,\u0022datetime\u0022:\u00222022-08-04T17:18:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-180}",
  "lastAlarm": "{\u0022code\u0022:816,\u0022datetime\u0022:\u00222022-08-04T17:08:52.000-00:00\u0022,\u0022type\u0022:\u0022ALERT\u0022,\u0022flash\u0022:false,\u0022instanceId\u0022:2801,\u0022messageId\u0022:\u0022BC_SID_HIGH_SG_CHECK_BG\u0022,\u0022sg\u0022:200,\u0022pumpDeliverySuspendState\u0022:false,\u0022referenceGUID\u0022:\u002211311ae5-c722-3e88-8f44-9cdcaad431e5\u0022,\u0022kind\u0022:\u0022ALERT\u0022,\u0022version\u0022:1}",
  "activeInsulin": "{\u0022amount\u0022:3.2,\u0022datetime\u0022:\u00222022-08-04T17:21:01.956Z\u0022,\u0022precision\u0022:\u0022TENTHS\u0022,\u0022kind\u0022:\u0022Insulin\u0022,\u0022version\u0022:1}",
  "sgs": "[{\u0022sg\u0022:247,\u0022datetime\u0022:\u00222022-08-03T17:23:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-86280},{\u0022sg\u0022:251,\u0022datetime\u0022:\u00222022-08-03T17:28:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-85980},{\u0022sg\u0022:250,\u0022datetime\u0022:\u00222022-08-03T17:33:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-85680},{\u0022sg\u0022:245,\u0022datetime\u0022:\u00222022-08-03T17:38:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-85380},{\u0022sg\u0022:238,\u0022datetime\u0022:\u00222022-08-03T17:43:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-85080},{\u0022sg\u0022:233,\u0022datetime\u0022:\u00222022-08-03T17:48:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-84780},{\u0022sg\u0022:234,\u0022datetime\u0022:\u00222022-08-03T17:53:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-84480},{\u0022sg\u0022:237,\u0022datetime\u0022:\u00222022-08-03T17:58:01.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-84180},{\u0022sg\u0022:237,\u0022datetime\u0022:\u00222022-08-03T18:03:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-83881},{\u0022sg\u0022:236,\u0022datetime\u0022:\u00222022-08-03T18:08:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-83581},{\u0022sg\u0022:234,\u0022datetime\u0022:\u00222022-08-03T18:13:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-83281},{\u0022sg\u0022:230,\u0022datetime\u0022:\u00222022-08-03T18:18:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-82981},{\u002 [... cut ...]
  "aboveHyperLimit": "82",
  "timeInRange": "18",
  "pumpCommunicationState": "True",
  "gstCommunicationState": "True",
  "gstBatteryLevel": "55",
  "lastConduitDateTime": "04/08/2022 17:20:42\u002B02:00",
  "maxAutoBasalRate": "0.9821",
  "maxBolusAmount": "16.9",
  "sensorDurationMinutes": "3306",
  "timeToNextCalibrationMinutes": "321",
  "clientTimeZoneName": "Central European Summer Time",
  "sgBelowLimit": "50",
  "averageSGFloat": "221.68231",
  "calFreeSensor": "False",
  "finalCalibration": "False"
}

paul1956, 2022-08-07:
@Komle92 
Fixed in [CareLinkv3.5.0.0x64](https://github.com/paul1956/CareLink/releases/tag/3.5.0.0)

Please open new issue if the issue persists

closed 2022-08-07 by paul1956