paul1956/CareLink #10: [Feature Request]: Replace mini display with taskbar GC reading.

Krikke99, 2022-08-01:
- Latest glucose reading in taskbar. 
- Color coded glucose numbers set to your ranges.
- Get time of reading, trend and active insulin on hover.

Krikke99, 2022-08-01:
[Discuss]
i'm trying to get the SG in the system tray, but it looks terrible.

[image]

i can't find a suitable font and can only use 16x16 pixels icon.

`Public Sub CreateTextIcon(str As String)
        Dim fontToUse As Font = New Font("Trebuchet MS", 10, FontStyle.Regular, GraphicsUnit.Pixel)
        Dim color As Color = Color.White
        Dim bgColor As Color = Color.Transparent
        Dim brushToUse As Brush = New SolidBrush(color)
        Dim bitmapText As Bitmap = New Bitmap(16, 16)
        Dim g As Graphics = System.Drawing.Graphics.FromImage(bitmapText)
        g.Clear(bgColor)
        g.TextRenderingHint = System.Drawing.Text.TextRenderingHint.SingleBitPerPixelGridFit
        g.DrawString(str, fontToUse, brushToUse, -2, 0)
        Dim hIcon As IntPtr = (bitmapText.GetHicon())
        Me.NotifyIcon1.Icon = System.Drawing.Icon.FromHandle(hIcon)
    End Sub`

closed 2022-08-03 by Krikke99

paul1956, 2022-08-03:
@Krikke99  Looks great, I merged.