paul1956/CareLink #9: change unit

Fishbaits, 2022-08-01:
Is there a way to change unit from mg/dL to mmol/L?

paul1956, 2022-08-03:
@Fishbaits  The graph mess up is related to date issue. It should automatically pick up units data from your pump. If you can upload the CareLink*.json file from your documents folder I can look into it further.

The json file is readable text and personal data is removed if you save as snapshot from the Start Up Menu of the app.

There is a beta version you can try under https://github.com/paul1956/CareLink/files/9193835/CareLinkv3.4.2.4x64.zip which fixes the graph but looking at it I have some precision issues I need to address.

Fishbaits, 2022-08-03:
When I try to run this new version, it says so:
[image]
And then quits, no matter if I click OK or Cancel

paul1956, 2022-08-03:
@Fishbaits thanks. I will upload a trial fix tomorrow.

This information will help.

paul1956, 2022-08-07:
@Fishbaits It looks like the pump stores and reports everything in mg/dl and display software scales it. So all the data is there but off the top of the graph. It is going to take some time to scale everything. I was going to post something that works today but it would be in US units. I will work on fixing the units next week.

paul1956, 2022-08-07:
@Fishbaits 
Fixed in [CareLinkv3.5.0.0x64](https://github.com/paul1956/CareLink/releases/tag/3.5.0.0)

Please open new issue if the issue persists

closed 2022-08-07 by paul1956