paul1956/CareLink #12: Datetime notation incorrect

Krikke99, 2022-08-03:
Datetime notation in the main screen is incorrect:

[image]

paul1956, 2022-08-03:
@Krikke99 this is actually a string that shows exactly what the pump has internally. I am working on an update this week that will also show the time in your time zone. The JSON Lib I was using was converting date strings to local time before I ever saw them.  I don't have the raw pump data that I display and I am working to displaying it in your computers native time format and time zone. so you could look at someone's data and it would make sense I am not there yet. If you can tell me what you would like to see there please open another issue or comment on this one.

Krikke99, 2022-08-03:
@paul1956 i just looked at a json file i get from a python program, the timezone is correct. Only the notation is not.
"lastSensorTSAsString": "2022-08-03T14:51:00.000Z"
I don't know how the notation is in other regions, in Belgium it looks like this: 03-08-2022 14:51

These are my windows settings:

[image]

paul1956, 2022-08-03:
That is a string directly from pump. It's not a Date but am thinking it's confusing and will eventually convert all the dates to local culture.

paul1956, 2022-08-07:
Fixed in [CareLinkv3.5.0.0x64](https://github.com/paul1956/CareLink/releases/tag/3.5.0.0)

Please open new issue if the issue persists

closed 2022-08-07 by paul1956