paul1956/CareLink #14: Still not working...

Fishbaits, 2022-08-08:
Hi Paul

Thank you for all the effort you are and have been putting into this but there are still issues.
When I run the program, in about 1 minute I get error message of Unhandled exception.
[image]
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/9279811/CareLinkErrorReport.en-US.txt)

paul1956, 2022-08-08:
@Fishbaits it looks like bad data from CareLink that I am not dealing with. I will look at it late today. I assume if you reset it will fail immediately and probably clear itself in 24 hours. I will know more when I look at your error report which does reproduce the issue.

paul1956, 2022-08-08:
Please try this early version and see if it fixes your issue.
It looks like it happens when your pump is disconnected from phone and because of the way I need to deal with international units instead of just ignoring it the app crashes
[CareLinkv3.5.0.3x64.zip](https://github.com/paul1956/CareLink/files/9285562/CareLinkv3.5.0.3x64.zip)
.

Fishbaits, 2022-08-10:
Hi

The CareLink3.5.0.5x64 was the latest by now and this one works!
[image]
The graph is there, everything is visible, great work!!
Thank you so very much!

paul1956, 2022-08-10:
The graph looks odd to me I don't understand the green lines, can you send me the saved data file? There is a newer version that fixes the 154 on the display above. I will leave this open if you want to use it to send me the file or you can close it.

Fishbaits, 2022-08-10:
Hi
The latest works also ubt I do not see any difference really. Graph looks the same.
[image]
I get a graph when I look at Running Active Insulin tab.
[image]

But what I personally miss the most is the speed of change. On big screen I cannot see it, on mini display it is there but numbers are in wrong unit again.
[image]

Fishbaits, 2022-08-10:
Where is this saved data file?
Everything in the Carelink folder has the same date and time, so I guess it is somewhere else?
Tried to do the Snapshot Save too but no clue where that went...

paul1956, 2022-08-10:
The working files are all saved in My Documents. I didn't write the mini display but will work on getting it fixes

paul1956, 2022-08-10:
@Fishbaits 
These are what the graphs are supposed to look like
[image]
[image]

Fishbaits, 2022-08-11:
Ok, these do look a bit different :)
One cause might also be, that I am running Win11.
[CareLinkLastDownload(et-EE).zip](https://github.com/paul1956/CareLink/files/9306898/CareLinkLastDownload.et-EE.zip)

paul1956, 2022-08-11:
Also on Windows 11. I fixed the 22 already. I will look into the zip tomorrow. This is what it should look like with European Data
[image]
[image]

paul1956, 2022-08-12:
@Fishbaits The reason the graphs look like they do is there is very little data. If you look at the file (you can open in any text editor or Visual Studio will decode it), most of the values are NaN. I don't know why but the program is working correctly. Even the bgUnit data is missing. The red values in the image below are the green lines on your graph, I don't what what to do with this report. I am curious what the phone app looked like when this happened. The latest version I will post soon can deal with all the NaN values so it will not crash. You can then look at the raw data and maybe see what the issue might be.
[image]

paul1956, 2022-08-12:
@Fishbaits Try this it works on my system with your data.
[CareLink3.5.1.0x64.zip](https://github.com/paul1956/CareLink/files/9329571/CareLink3.5.1.0x64.zip)

paul1956, 2022-08-12:
@Fishbaits Issue finally found I was rounding small numbers with decimal points in some cultures
Here is the fixed version
[image]
[CareLink3.5.1.0x64.zip](https://github.com/paul1956/CareLink/files/9329862/CareLink3.5.1.0x64.zip)

closed 2022-08-12 by paul1956