paul1956/CareLink #15: crashes on every attempt

Fishbaits, 2022-08-15:
Hi Paul

Again, thank you for all your effort.
Sorry, I am not a programmer and cannot help you much.
At moment the latest version crashes on every attempt.
It does the login but probably graph drawing causes the crash.
And yes, we have comma as decimal separator, not dot. So a number looks like 5,43
[image]
[CareLinkErrorReport.zip](https://github.com/paul1956/CareLink/files/9336277/CareLinkErrorReport.zip)

paul1956, 2022-08-16:
@Fishbaits This is the text file you sent running on my system and everything looks perfect with commas where they belong and 24-hour time format
[image]
There is an update here to prevent the overflow, if it doesn't work, I have one more thing to try then without running a debugger on your system when it crashes, I have no way to fix further.
[CareLink3.5.1.7x64.zip](https://github.com/paul1956/CareLink/files/9347411/CareLink3.5.1.7x64.zip)

Fishbaits, 2022-08-16:
Still crashes.
Exception: Input string was not in a correct format.
[image]
[CareLinkErrorReport(en-US)3.txt](https://github.com/paul1956/CareLink/files/9351637/CareLinkErrorReport.en-US.3.txt)

Fishbaits, 2022-08-16:
[image]

paul1956, 2022-08-16:
It's interesting that the error report is en-US but all the data is et-EE your local format.

paul1956, 2022-08-17:
This version has a speciic fix for the above error even though is should not happen
[CareLink3.5.1.8x64.zip](https://github.com/paul1956/CareLink/files/9356141/CareLink3.5.1.8x64.zip)

Fishbaits, 2022-08-17:
Still throws exception...
[image]
[CareLinkErrorReport(en-US)5.txt](https://github.com/paul1956/CareLink/files/9358259/CareLinkErrorReport.en-US.5.txt)

paul1956, 2022-08-19:
What language is you Windows running in and your pump? Would you like to try a remote debug session at a time that works for you. We would install a free copy of Visual Studio and connect with Chrome remote desktop.

paul1956, 2022-08-20:
@Fishbaits your last report actually shows a different error, and I found a few more places I was parsing wrong. Please try this one.
[CareLink3.5.1.9x64.zip](https://github.com/paul1956/CareLink/files/9386546/CareLink3.5.1.9x64.zip)
.

Fishbaits, 2022-08-20:
paul1956, there is a clear change in error messages now but it still crashed.
[image]
[CL.zip](https://github.com/paul1956/CareLink/files/9386620/CL.zip)
I am running Windows in Estonian language but I also tried it on a pc with Win10 64bit Pro Eng and got the same result.

paul1956, 2022-08-21:
@Fishbaits thanks for you supper getting this fixed. This version has some more error reporting and some fixes. Please try.
[CareLink3.5.1.10x64.zip](https://github.com/paul1956/CareLink/files/9388211/CareLink3.5.1.10x64.zip)

Fishbaits, 2022-08-21:
There is some visible progress but still an exception appered.
[image]
[image]
[CareLinkErrorReport(en-US)7.txt](https://github.com/paul1956/CareLink/files/9389806/CareLinkErrorReport.en-US.7.txt)

paul1956, 2022-08-22:
@Fishbaits This will probably still crash but should produce a better report.
[CareLink3.5.1.12x64.zip](https://github.com/paul1956/CareLink/files/9391841/CareLink3.5.1.12x64.zip)
t

Fishbaits, 2022-08-22:
Yes, it crashed.
[image]
[CL.zip](https://github.com/paul1956/CareLink/files/9392982/CL.zip)

paul1956, 2022-08-22:
Thanks for all the help. That helped a lot, I never would have found the issue. It may not be the only one. Please try this.
[CareLink3.5.1.12x64a.zip](https://github.com/paul1956/CareLink/files/9393518/CareLink3.5.1.12x64a.zip)
.

Fishbaits, 2022-08-22:
And we have a winner!
It seems to work all properly now, no exceptions or anything.
Graph is displayed, all seems ok. Great work!!
[image]

paul1956, 2022-08-22:
Thanks for all your help.

paul1956, 2022-08-22:
I am going to close

closed 2022-08-22 by paul1956