paul1956/CareLink #17: Login failed

Krikke99, 2022-08-24:
Login failed if the application is minimized for a while.

[image]

paul1956, 2022-08-25:
@Krikke99 CareLink has been very flakey especially in Europe due to DNS issues. If I fail twice, I just give up. I will look into retrying. CareLink has a 3 try and then you get kicked out for 15 minutes and I don't want to hit that. Open to any suggestions.

paul1956, 2022-08-26:
@Krikke99  I think this is fixed in 3.5.3.0 which I am posting now.

Krikke99, 2022-08-26:
@paul1956 the problem remains in version 3.5.3.0

paul1956, 2022-08-27:
@Krikke99 I found something in the retry logic that may cause it. I will post a fix Sunday.

paul1956, 2022-09-05:
@Krikke99 Fixed in CareLink3.5.4.6x64. Please close if you agree.

Krikke99, 2022-09-07:
@paul1956 now the application does not update anymore.
After 1 minute:
[image]
After 5 minutes:
[image]

paul1956, 2022-09-08:
I have rewritten the login logic in CareLink3.5.5.0x64 Let me know if it fixes the problem

Krikke99, 2022-09-08:
@paul1956 
In in CareLink3.5.5.0x64 no error message anymore , but also no updates.

[image]
[image]

paul1956, 2022-09-08:
@Krikke99 I have put in 2 for fixed for this issue in CareLink3.5.5.1x64 hopefully they work for you.

paul1956, 2022-09-08:
@Krikke99 I look like Carelink is getting more aggressive with requiring reauthorizing. I was only doing it when the previous one expired every 24 hours. In the past retrying just worked not I need to relog in. Hopefully this works.

closed 2022-09-19 by Krikke99