paul1956/CareLink #18: Object reference not set to an instance of an object. exception in AllTabPagesUpdate at 1847

Zxquad, 2022-09-08:
Hi, 
Wanna try your app but ;)

   at CareLink.Form1.UpdateActiveInsulinChart(String memberName, Int32 sourceLineNumber) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 1961
   at CareLink.Form1.AllTabPagesUpdate() in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 1847
   at CareLink.Form1.Form1_Shown(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 258
   at System.Windows.Forms.Form.OnShown(EventArgs e)
   at System.Windows.Forms.Form.CallShownEvent()
   at System.Windows.Forms.Control.InvokeMarshaledCallbackDo(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)

In Belgium on win10

Thx for your work

paul1956, 2022-09-09:
@Zxquad Try 3.5.5.2 and if it does not fix your issue,  can you attach the Error File its under in your documents folder?

paul1956, 2022-09-09:
@Zxquad delete or edit CareLink.Csv  in my documents to change saved options when the app crashes but I think your crash is solved in latest update.

Zxquad, 2022-09-09:
@paul1956 Working just fine now with 3.5.5.2 ;)
thx for fixing the issue ...

closed 2022-09-09 by paul1956