paul1956/CareLink #19: Timezone issue... Central European Summer Time

HJADR, 2022-09-12:
[image]

I am not sure what to select.  Selected cancel 2 times, but I see no created issue. And well... I am just not sure, might be a problem English not beeing my native language...

paul1956, 2022-09-13:
You need to say Yes and it will stop prompting you or No and it will keep asking. In your case you can send this file CareLinkLastDownload(en-US).json  in you Documents folder. (en-US) will be something else as "en" is English and "US" is my country.

HJADR, 2022-09-13:
[CareLinkLastDownload(nl-NL).zip](https://github.com/paul1956/CareLink/files/9558092/CareLinkLastDownload.nl-NL.zip)

Here you go

paul1956, 2022-09-14:
Timezone issue should also be fixed in 3.5.5.3

closed 2022-10-01 by paul1956