paul1956/CareLink #20: SG trend not showing correctly

HJADR, 2022-09-12:
[image]
[image]

It is showing +5 and not +0,05, the BG unit we use here is mmol/L, thats most likely why it is off. not a real big issue.. but hey.

paul1956, 2022-09-13:
If you post your version of  CareLinkLastDownload(en-US).json here, I can fix this easily.

HJADR, 2022-09-13:
[CareLinkLastDownload(nl-NL).zip](https://github.com/paul1956/CareLink/files/9558096/CareLinkLastDownload.nl-NL.zip)

Here you go. :)

paul1956, 2022-09-14:
@HJADR actually +5 is the default when the value is unchanged as a sample. This issue fixed in 3.5.5.3
[image]
Please close if you agree.

HJADR, 2022-09-15:
I am not sure yet. I get this error details>

`   at CareLink.CareLinkUserDataList.ContainsKey(String key) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Records\UserDataRecords\CareLinkUserDataList.vb:line 191
   at CareLink.LoginForm1.LoginForm1_Load(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\LoginForm1.vb:line 54
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Form.OnCreateControl()
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.ContainerControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WmShowWindow(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)`

paul1956, 2022-09-15:
Sorry, that is a separate issue that was cause by supporting Multiuser. It should be fixed in version I will post today for everyone.
[CareLink5.5.6.0x64.zip](https://github.com/paul1956/CareLink/files/9579010/CareLink5.5.6.0x64.zip)

closed 2022-10-01 by paul1956