paul1956/CareLink #21: Build errors

Krikke99, 2022-09-26:
i get a lot of build errors in the latest version.

[image]

Krikke99, 2022-09-26:
Also the latest release doesn't work anymore.

[image]

paul1956, 2022-09-27:
@Krikke99 You need to Clone https://github.com/kirsan31/winforms-datavisualization its the new improved DataVisualization package but it is not available as a NewGet yet. Issue is fixed it was an initialization error caused by partial implementation of editor for the config file. It should be fixed in 3.5.6.6.

Krikke99, 2022-09-28:
@paul1956 build errors are gone, but both build and release now give this error

[image]

paul1956, 2022-09-29:
Try deleting Carelink.csv is myDocuments I think it is corrupt. I put in some fixes for that in a version I am working on but not ready to release. Otherwise, can you set a break on Link 164 and tell me what past of the line is Nothing.

Krikke99, 2022-09-30:
After deleting the file everything works again.