paul1956/CareLink #25: Unable to login

nfontan, 2022-10-17:
Unable to login on Latin America Server (Argentina).
I tried with 2 differents account and computers.

Account:
nfontan (patient)
nfontanc (care partner)
[image]
[image]

No logs on document folder

nfontan, 2022-10-17:
After selecting the europe server the following errors are received

[https://pastebin.com/YbCu2Mn3](https://pastebin.com/YbCu2Mn3?fbclid=IwAR1X6WjDPMu3qqmmT1dV_ppvukuWN_wAARRIPOHFIxpgm2-lEGRESZmZD6o)
[https://pastebin.com/Xm5cuj4r](https://pastebin.com/Xm5cuj4r?fbclid=IwAR0kgn5TNYwkZ5X5e4dxsYjYlPxlGJaLsc-lJ59Tzs3uyQjZHO28Sg43M9M)
[https://pastebin.com/r40dMA1A](https://pastebin.com/r40dMA1A?fbclid=IwAR0ZFFbP0dvkb9ZZfPNSOJCTTi17osXzeFug9O_ahMQBpIqUNhH7DFxuDzM)
[https://pastebin.com/cUDU7YHG](https://pastebin.com/cUDU7YHG?fbclid=IwAR0zTxv5CfraXJvWfNbFnXc8OQAGwxf-F5ICCVzBJGfo81AKFv3Fq4kUIrI)
[https://pastebin.com/Yhr9bKut](https://pastebin.com/Yhr9bKut?fbclid=IwAR0CTEeVkQJ4mCy-4C2XVfOD4rEHuXJ-Qf3WKAvcgIDcIeNePhgh0-1giJM)

I think the problem is the timezone
"clientTimeZoneName": "Argentina Standard Time",

paul1956, 2022-10-18:
@nfontan the time zone issue is I can't automatically map "Argentina Standard Time" to anything known. All the know zones at GMT-3 are listed below is one of the correct for you?
[image]

paul1956, 2022-10-18:
@nfontan I have solved many of the issue, it appears you are in MM and I have done no testing in that mode. It will take a week or so to address all the issues. Thanks for all the reports, below is your last error file now working.
[image]

nfontan, 2022-10-18:
> @nfontan I have solved many of the issue, it appears you are in MM and I have done no testing in that mode. It will take a week or so to address all the issues. Thanks for all the reports, below is your last error file now working.
Thank you for this great work!

I am currently in MM because I migrated from a 640 to a 780 a week ago and my educator told me that before going to manual I should be like this for a few days, but I think that next week I will go into auto mode. Anyway if you need me to try something I will be available for you

paul1956, 2022-10-19:
[CareLink3.5.6.11x64.zip](https://github.com/paul1956/CareLink/files/9816734/CareLink3.5.6.11x64.zip)
@nfontan please try this, I am not ready to release it yet, but I believe all your issues are fixed.
Please let me know.

paul1956, 2022-10-19:
@nfontan I don't have any user in MM so I am glad I captured your error file it pointed out server issues. In the future you can just drop the file here or use pasteBin but the files are just text

nfontan, 2022-10-19:
> @nfontan I don't have any user in MM so I am glad I captured your error file it pointed out server issues. In the future you can just drop the file here or use pasteBin but the files are just text

Hi Paul,
This version works great!

Just a small detail, in the drop-down menu of the login where the area and the country appear, Argentina continues to give an error and I have to choose Spain. Could you refer South America to the same European server?

Thanks again!

paul1956, 2022-10-19:
@paul1956 I do use the same server for South America and Spain with the version I posted above. I am looking at the issue as I see it too.

nfontan, 2022-10-19:
> @paul1956 I do use the same server for South America and Spain with the version I posted above. I am looking at the issue as I see it too.

Paul, At the moment I am testing it on my home pc and the login works correctly with both countries. Previously I had tested it on the office pc. Capturing the dns traffic with my pihole server I saw that it generated two queries, the first went to mdtlogin.medtronic.com and the second to carelink.minimed.eu

paul1956, 2022-10-20:
@nfontan since you seem similar with code would you like to try a joint debug session this weekend or next week? Everything you need is available free.

nfontan, 2022-10-20:
> @nfontan since you seem similar with code would you like to try a joint debug session this weekend or next week? Everything you need is available free.

Yes, no problem!

Only that my level of dialogue in English is limited, but we can chat

paul1956, 2022-10-21:
@nfontan  you can find me on Gitter as @paul1956 and we can chat privately.

I am available next 3 days after 3:00 PM Pacific Time, just pick a time that works for you

closed 2022-10-24 by paul1956