paul1956/CareLink #26: Problems after first start

Bernsti76, 2022-10-23:
[image]

I get this after the first start at every next start of the programme.

paul1956, 2022-10-24:
@Bernsti76 I need the files it creates under My Documents starting with CareLink.

paul1956, 2022-10-24:
@Bernsti76 This should fix your issue, but it is not ready for release you may get some popup boxes. You will need to delete careLink.csv it is definitely corrupted,
[CareLink3.5.7.0x64.zip](https://github.com/paul1956/CareLink/files/9850366/CareLink3.5.7.0x64.zip)

Bernsti76, 2022-10-25:
> @Bernsti76 You can also try deleting CareLink.csv. Sometimes it gets corrupt. I found the reason and need to update my code.

Thank you so much! After deleting CareLink.csv everything is working again!

paul1956, 2022-10-26:
@Bernsti76 I need to figure out why is gets corrupt, I will close this.

closed 2022-10-26 by paul1956