paul1956/CareLink #27: Login error - Sequence contains no elements

mattimart, 2022-11-09:
[image]

nfontan, 2022-11-09:
> [image]

Please, attach the CareLinkErrorReport(en-US)xx.txt locate on c:\users\youruser\documents

paul1956, 2022-11-10:
@mattimart @nfontan I need the error report to look at this.

paul1956, 2022-11-10:
Just from looking at the error, it looks like server is returning no data for some of the fields and I don't guard against it. If I had error report, I could verify it.

paul1956, 2022-11-10:
[CareLink3.5.7.6x64.zip](https://github.com/paul1956/CareLink/files/9977489/CareLink3.5.7.6x64.zip)
Try this. these errors are caused by CareLink server or frontend down.

paul1956, 2022-11-11:
@mattimart @nfontan Fixed in CareLink3.5.7.7x64. Close if you agree.

closed 2022-11-11 by mattimart