paul1956/CareLink #31: Doesnt add up

JJP780G, 2022-11-25:
[image]

Doing the maths the total insulin should be 16.1 units not 13.9

I am using the previous version as the new version shuts down while trying to enter the username. 

Many Thanks

paul1956, 2022-11-26:
Delete CareLink.csv that should address the shutdown issue.

 

I need the data file in My Documents to look at the Math. What pump?

 

 <https://user-images.githubusercontent.com/91048750/203894314-2878c7b5-7e6e-4e69-8226-d59f90d7e4d4.png>

paul1956, 2022-11-28:
@JJP780G 
Math error fixed in CareLink3.5.7.9x64

Please close of you agree or attach CareLinkLastDownload(en-US).json file. name will vary based on country. It is a text file you can review all private information should be deleted.

closed 2022-12-14 by paul1956