paul1956/CareLink #32: Error login and update

alexklit, 2022-12-10:
Hi when I try to login the app crashes because it can't recognize the regional format for Greece (see attached file), when I change the regional format to English (United Kingdom) it works properly for some time and then it appears the message : The format of value 'application/json; charset=utf-8' is invalid and the update stops
[CareLinkErrorReport(en-US)5.txt](https://github.com/paul1956/CareLink/files/10200102/CareLinkErrorReport.en-US.5.txt)
[carelink errors.docx](https://github.com/paul1956/CareLink/files/10200107/carelink.errors.docx)

paul1956, 2022-12-12:
Sorry for the long delay.  I am mostly offline with no access to GitHub until 12/17.

 

The second issue ('application/json; charset=utf-8') is fixed but I can’t upload the fix until 12/17. If you go to the previous version also on GitHub that version does not have the error. It has a few 780G bugs around where total daily insulin use has some double counting and the time is wrong in one place but generally works well.

 

I will try to fix the issue with Region issue on Saturday but I don’t access the two files until Saturday Morning and I lose internet again Saturday evening (I am on a ship having just left Brazil heading for Florida, then leaving for Panama where I again will have unrestructured internet for just a few hours. 

 

If you can email both files as attachment to ***@***.*** ***@***.***>  I can work on them this week.

 

When I get your email I will email you an experimental version with the fix for the communication error and new features on the AIT tab and well add Greece fix and send that in a day or so.

 

If you don’t want to share email address that’s fine I will just upload new version when I return to Florida.

paul1956, 2022-12-14:
@alexklit I believe this and a lot of other issues are fixed in CareLink3.5.7.16x64.zip.

Close if you agree

alexklit, 2022-12-15:
Hi Paul, thank you for your immediate response despite your absence on a
trip. My feedback is that the second issue with  " The format of value
'application/json;" is resolved, the only issue that remains is the login
with region "Greece", I send you again the error file.
Best regards.

Klitorakis Alexandros

--- Start of Exception ---
String '14/12/2022 10:29:45 πμ' with key = dateTime from dgvCellFormatting line 24 was not recognized as a valid DateTime in any supported culture.
--- End of Exception ---
--- Start of stack trace ---
   at CareLink.DateTimeExtensions.ParseDate(String dateAsString, String key, String memberName, Int32 sourceLineNumber) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Extensions\DateTimeExtensions.vb:line 196
   at CareLink.DataGridViewExtensions.dgvCellFormatting(DataGridView dgv, DataGridViewCellFormattingEventArgs& e, String key) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Extensions\DataGridViewExtensions.vb:line 24
   at CareLink.Form1.DataGridViewAutoBasalDelivery_CellFormatting(Object sender, DataGridViewCellFormattingEventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 817
   at System.Windows.Forms.DataGridViewCell.GetFormattedValue(Object value, Int32 rowIndex, DataGridViewCellStyle& cellStyle, TypeConverter valueTypeConverter, TypeConverter formattedValueTypeConverter, DataGridViewDataErrorContexts context)
   at System.Windows.Forms.DataGridViewTextBoxCell.GetPreferredSize(Graphics graphics, DataGridViewCellStyle cellStyle, Int32 rowIndex, Size constraintSize)
   at System.Windows.Forms.DataGridViewCell.GetPreferredWidth(Int32 rowIndex, Int32 height)
   at System.Windows.Forms.DataGridViewColumn.GetPreferredWidth(DataGridViewAutoSizeColumnMode autoSizeColumnMode, Boolean fixedHeight)
   at System.Windows.Forms.DataGridView.AutoResizeColumnInternal(Int32 columnIndex, DataGridViewAutoSizeColumnCriteriaInternal autoSizeColumnCriteriaInternal, Boolean fixedHeight)
   at System.Windows.Forms.DataGridView.AutoResizeAllVisibleColumnsInternal(DataGridViewAutoSizeColumnCriteriaInternal autoSizeColumnCriteriaFilter, Boolean fixedHeight)
   at System.Windows.Forms.DataGridView.OnAddedRow_PostNotification(Int32 rowIndex)
   at System.Windows.Forms.DataGridViewRowCollection.OnCollectionChanged_PostNotification(CollectionChangeAction cca, Int32 rowIndex, Int32 rowCount, DataGridViewRow dataGridViewRow, Boolean changeIsDeletion, Boolean changeIsInsertion, Boolean recreateNewRow, Point newCurrentCell)
   at System.Windows.Forms.DataGridViewRowCollection.OnCollectionChanged(CollectionChangeEventArgs e, Int32 rowIndex, Int32 rowCount)
   at System.Windows.Forms.DataGridViewRowCollection.AddInternal(DataGridViewRow dataGridViewRow)
   at System.Windows.Forms.DataGridView.RefreshRows(Boolean scrollIntoView)
   at System.Windows.Forms.DataGridView.OnDataSourceChanged(EventArgs e)
   at CareLink.DataGridViewHelper.DisplayDataTableInDGV(TableLayoutPanel realPanel, DataGridView dGV, DataTable table, ItemIndexs rowIndex) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Helpers\DataGridViewHelper.vb:line 42
   at CareLink.UpdateTabHelpers.UpdateMarkerTabs(Form1 MainForm) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Helpers\UpdateTabHelpers.vb:line 78
   at CareLink.Form1.UpdateAllTabPages() in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 2111
   at CareLink.Form1.Form1_Shown(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\Form1.vb:line 179
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
{
  "lastSensorTS": "0",
  "medicalDeviceTimeAsString": "2022-12-15T10:24:51.216-00:00",
  "lastSensorTSAsString": "2022-12-15T10:20:00.000Z",
  "kind": "BLENGP",
  "version": "2",
  "pumpModelNumber": "MMT-1886",
  "currentServerTime": "1671092893779",
  "lastConduitTime": "0",
  "lastConduitUpdateServerTime": "1671092718590",
  "lastMedicalDeviceDataUpdateServerTime": "1671092718590",
  "firstName": "First",
  "lastName": "Last",
  "conduitSerialNumber": "BCB478-BB10-955A-EC47-0000-0000-0000-0000",
  "conduitBatteryLevel": "85",
  "conduitBatteryStatus": "FULL",
  "conduitInRange": "True",
  "conduitMedicalDeviceInRange": "True",
  "conduitSensorInRange": "True",
  "medicalDeviceFamily": "NGP",
  "sensorState": "NO_ERROR_MESSAGE",
  "medicalDeviceSerialNumber": "NG1234567H",
  "medicalDeviceTime": "0",
  "sMedicalDeviceTime": "2022-12-15T10:24:51.216-00:00",
  "reservoirLevelPercent": "13",
  "reservoirAmount": "300",
  "reservoirRemainingUnits": "40.1",
  "medicalDeviceBatteryLevelPercent": "75",
  "sensorDurationHours": "77",
  "timeToNextCalibHours": "3",
  "calibStatus": "LESS_THAN_SIX_HRS",
  "bgUnits": "MGDL",
  "timeFormat": "HR_24",
  "lastSensorTime": "0",
  "sLastSensorTime": "2022-12-15T10:20:00.000Z",
  "medicalDeviceSuspended": "False",
  "lastSGTrend": "NONE",
  "lastSG": "{\u0022sg\u0022:198,\u0022datetime\u0022:\u00222022-12-15T10:20:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-291}",
[... cut ...] \u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-16191},{\u0022sg\u0022:110,\u0022datetime\u0022:\u00222022-12-15T06:00:00.000Z\u0022,\u0022timeChange\u0022:false,\u0022sensorState\u0022:\u0022NO_ERROR_MESSAGE\u0022,\u0022kind\u0022:\u0022SG\u0022,\u0022version\u0022:1,\u0022relativeOffset\u0022:-15891},{\u0022sg\u0022:102,\u0022datetime\u0022:\u00222022-12-15T06:05:00.000Z\u0022,\u0022timeChange\u0022:false,\u

paul1956, 2022-12-15:
The file is not complete, the error file below is not complete. I will have to look at it Saturday when I have full internet.

Importance: High

alexklit, 2022-12-15:
Ok, no problem.

paul1956, 2022-12-16:
I think the issue is the Greek characters πμ in the time and the program does not know what to do with them.

paul1956, 2022-12-17:
@alexklit The problem is the 2 digit code for Greece GR, which gives me en-GR is not one of the 200 known country codes. I assume GR isn't correct and I would translate it if I know what was correct. I will try to send you a tiny test program that can output your internal code and then I can fix this issue. It will probably be 10 days, but I will post it here.

paul1956, 2022-12-23:
@alexklit please try this and make sure everything is working. I will be online today until 10 PM EST. and again tomorrow,
[CareLink3.5.7.18x64.zip](https://github.com/paul1956/CareLink/files/10295473/CareLink3.5.7.18x64.zip)

If everything is working, please close.

alexklit, 2022-12-23:
you are amazing, happy holidays

paul1956, 2022-12-23:
@alexklit Happy Holidays to you and yours as well

closed 2022-12-23 by paul1956