paul1956/CareLink #34: Object reference not set to an instance of an object.

errezeta67, 2023-01-24:
at CareLink.CareLinkUserDataList.get_Item(String itemName) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Records\UserDataRecords\CareLinkUserDataList.vb:line 100
   at CareLink.LoginForm1.LoginForm1_Load(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\LoginForm1.vb:line 55
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)

paul1956, 2023-01-25:
Delete a "careLink.csv" under MyDocuments, and let me know if issue is not resolved

errezeta67, 2023-01-25:
problem solved thanks

paul1956, 2023-01-25:
I happens when people upgrade, and I don't detect the format change. I am working on handling it better.

closed 2023-01-25 by paul1956

errezeta67, 2023-01-25:
good job