paul1956/CareLink #36: Object reference not set to an instance of an object.

triomen, 2023-01-29:
Hello,

I can't login to the application. Once I try to type my password, the application shows this error 
`--- Start of Exception ---
Object reference not set to an instance of an object.
--- End of Exception ---
--- Start of stack trace ---
   at CareLink.LoginForm1.UsernameTextBox_Leave(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\LoginForm1.vb:line 149
   at System.Windows.Forms.ContainerControl.UpdateFocusedControl()
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}`

I is the first time I opened the application. I reinstall .NET 7.0 to be sure, but the behavior stays the same. 
Windows version : Win11 22H2
[CareLinkErrorReport(en-US)2.txt](https://github.com/paul1956/CareLink/files/10531033/CareLinkErrorReport.en-US.2.txt)

paul1956, 2023-01-30:
@triomen Try deleting CareLink.csv in Documents folder that usually fixes the issue.

paul1956, 2023-01-30:
@triomen Version .23 should prevent the crash but deleting the file is a better solution. Close if either solution fixes your issue.

triomen, 2023-01-31:
Thanks ! Deleting CareLink.csv in Documents folder worked ! (Downloading last version didn't fix the issue)

closed 2023-01-31 by triomen

paul1956, 2023-01-31:
@triomen  I need someone to post a broken file so I can solve it for good.