paul1956/CareLink #38: Program fails to start

dimsmara, 2023-03-06:
--- Start of Exception ---
Key not found (Parameter 'Key')
--- End of Exception ---
--- Start of stack trace ---
   at CareLink.SummaryRecordHelpers.GetValue[T](List`1 l, String Key) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Records\Pump Records\SummaryRecordHelpers.vb:line 186
   at CareLink.Form1UpdateHelpers.Is770G() in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Helpers\UiHelpers\Form1UpdateHelpers.vb:line 32
   at CareLink.InitializeDialog.InitializeDialog_Load(Object sender, EventArgs e) in C:\Users\PaulM\Source\Repos\CareLink\src\CareLink\Forms\InitializeDialog.vb:line 189
   at System.Windows.Forms.Form.OnLoad(EventArgs e)
   at System.Windows.Forms.Control.CreateControl(Boolean fIgnoreVisible)
   at System.Windows.Forms.Control.CreateControl()
   at System.Windows.Forms.Control.WmShowWindow(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.ScrollableControl.WndProc(Message& m)
   at System.Windows.Forms.Form.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}

dimsmara, 2023-03-06:
[CareLinkErrorReport(en-US)1.txt](https://github.com/paul1956/CareLink/files/10902640/CareLinkErrorReport.en-US.1.txt)

paul1956, 2023-03-06:
@dimsmara thanks for the report, see if this fixes the issue.
[CareLink3.6.0.2x64.zip](https://github.com/paul1956/CareLink/files/10902899/CareLink3.6.0.2x64.zip)

dimsmara, 2023-03-06:
Thanks  a lot! yes, it is running. But I have another issue now. Please check also the attached picture. The basal and correction bars are too short. It was not like this in the previous edition (CareLink3.5.7.23x64). Also, is it any way to see when the cannula set needs change? Thank you very much and best wishes from Greece!
[image]

paul1956, 2023-03-07:
@dimsmara Can you open another issue on the short bars and I will need a data file (.json)  If you take Snapshot, it will remove any personal information.  I have not seen this with other users those values should scale.
[image]

I will close this issue as complete.

paul1956, 2023-03-07:
@dimsmara I don't believe there is any way to know how long cannula has been in or if it's a 3-day or Extended version, % reservoir is based on how many units it started with not out of 300U. All the raw pump data is should on the 23 tabs, if you see something that you find useful and want it displayed somewhere else or differently let me know is a separate issue.
[image]

[image]

You can navigate to most data tabs from summary and then back again to summary with 1 click. The last 5 tabs are user data from CareLink and not the pump.

closed 2023-03-07 by dimsmara