paul1956/CareLink #39: Short basal and correction bars

dimsmara, 2023-03-07:
Hi Paul, as I wrote yesterday, I have an issue with short basal and correction bars with the new version CareLink3.6.0.3x64. I tried to attached the required file _CareLinkLastDownload(el-GR).jso_ but was not allowed from the system. So I have uploaded to WeTrasnfer, here is the relative link [](https://we.tl/t-a9SnsvPltu) . Please advise if I have to upload it in a different way. Thank you.

paul1956, 2023-03-08:
@dimsmara
[image]

I believe this is fixed in 3.6.0.4. If it is not, delete your settings file MyDocuments/CareLink/Settings and then the yourName.json and restart application. Then make sure all values are entered. The settings file should look like below but with your values. If it doesn't delete your name and post here.

```
{
  "CarbRatios": [
    {
      "StartTime": "00:00:00",
      "EndTime": "08:00:00",
      "CarbRatio": 6
    },
    {
      "StartTime": "08:00:00",
      "EndTime": "00:00:00",
      "CarbRatio": 4.5
    }
  ],
  "CurrentTarget": 120,
  "InsulinRealAit": 3,
  "InsulinTypeName": "Lyumjev\u00AE",
  "PumpAit": 2.25,
  "UseAdvancedAitDecay": 1,
  "UserName": "TestFile"
}
```

dimsmara, 2023-03-11:
Hi, I just tried ver.CareLink3.6.1.0 but still the same issue with the short bars. That's the json file:

{
  "CarbRatios": [
    {
      "StartTime": "00:00:00",
      "EndTime": "00:00:00",
      "CarbRatio": 15
    }
  ],
  "CurrentTarget": 100,
  "InsulinRealAit": 4,
  "InsulinTypeName": "Novolog\u00AE",
  "PumpAit": 2.5,
  "UseAdvancedAitDecay": 1,
  "UserName": ""
}

paul1956, 2023-03-11:
@dimsmara see if this fixes it. Otherwise please post the lastdownload .json file here
[CareLink3.6.1.1x64.zip](https://github.com/paul1956/CareLink/files/10947577/CareLink3.6.1.1x64.zip)

paul1956, 2023-03-11:
@dimsmara 
I need CareLinkLastDownload(en-US).json where the "en-US" is unique to your system.

paul1956, 2023-03-11:
[CareLink3.6.1.2x64.zip](https://github.com/paul1956/CareLink/files/10948261/CareLink3.6.1.2x64.zip)

@dimsmara this should work better if not I still need the file. The issue is  round trip converting of Singles from languages the use a comma for decimal separator.

dimsmara, 2023-03-11:
Yes, now it is fine! Thank you a lot!

closed 2023-03-11 by dimsmara