paul1956/CareLink #40: Visual bug and "report a bug button"

nfontan, 2023-03-07:
the new version seems to have some sort of visual bug. The basal and bolus lines were too small.
[image]

An other error when clink on "report a bug"
[image]

my chrome is installed on C:\Program Files\Google\Chrome\Application
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/10912874/CareLinkErrorReport.en-US.txt)

are you still using gitter?

Thanks

paul1956, 2023-03-08:
Browser bug fixed in 3.6.0.4

Yes, still on Gitter using new version.

I am closing as visual bug is already reported.

closed 2023-03-08 by paul1956

paul1956, 2023-03-12:
@nfontan This is fixed in the current version it happened when comma is used was decimal separator which I mostly handled but I missed 1. It also has the ability to export the larger tables to Clipboard for a Formatted Excel.