paul1956/CareLink #41: version 3.6.1.2 crash

alexklit, 2023-03-21:
In the version 3.6.1.2 the app the app crash.
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/11025476/CareLinkErrorReport.en-US.txt)

paul1956, 2023-03-21:
@alexklit I don't know what to do with a date that looks like '20/3/2023 7:05:49 Ï€Î¼'

alexklit, 2023-03-21:
The application worked properly before the version 3.6 with the same
regional settings.
Regards
Klitorakis Alexandros

paul1956, 2023-03-21:
@alexklit sorry you are correct. I found the issue.
[CareLink3.6.2.0x64.zip](https://github.com/paul1956/CareLink/files/11027014/CareLink3.6.2.0x64.zip)
See if this fixes it. If not the error will be somewhere else and I will need another report.

alexklit, 2023-03-21:
Thank you, it works properly.
Regards
Klitorakis Alexandros

closed 2023-03-21 by alexklit