paul1956/CareLink #43: Widget not fully showing the numbers and change to glucose levels

Ryptahi, 2023-06-11:
[image]

I noticed that when I have it show blood glucose in mmol/L the decimal numbers do not show correctly. Also change to the blood glucose is shown as -1 when actual change was -0.60. This issue is most likely only with european way of using mmol/L vs mg/dL, but could it be possible to show actual changes in decimals as well, as it feels like a bug or oversight.

paul1956, 2023-06-12:
@Ryptahi Do you mean the "," because that comes from your local Windows settings. The fact that it doesn't fit is a real issue that I will look at. But I have no test sites that use mmol/L so it will be difficult to reproduce.

paul1956, 2023-06-12:
@Ryptahi please try this
[CareLink3.8.2.0x64.zip](https://github.com/paul1956/CareLink/files/11716846/CareLink3.8.2.0x64.zip)

Ryptahi, 2023-06-12:
> @Ryptahi please try this [CareLink3.8.2.0x64.zip](https://github.com/paul1956/CareLink/files/11716846/CareLink3.8.2.0x64.zip)

Thank you, now it is absolutely perfect!

paul1956, 2023-06-12:
Thanks I will update and release a new version today.

closed 2023-06-12 by paul1956