paul1956/CareLink #46: Nightscout / sponsorship

bewest, 2023-08-18:
Howdy, great work!  FWIW, I notice a lot of defensive warnings that I don't think are needed:
* Right to repair protects patient rights use third party software without violating terms of service.  In certain states it is illegal for manufacturers to claim this, and it is being [enforced on a federal](https://www.ftc.gov/business-guidance/blog/2022/07/ftc-announces-three-right-repair-cases-do-your-warranties-comply-law) basis by the FTC.
* The risk is low - this is may be considered Class I, low-risk, as it is read-only data.  In any case [T1Pal is registered](https://www.accessdata.fda.gov/scripts/cdrh/cfdocs/cfrl/rl.cfm?start_search=1&regnum=295535) with FDA as secondary display remote monitor and would be happy to add this carelink viewer as another registered device.

I'm very interested in promoting Nightscout connectivity, how would you feel about additional synchronizing to or from Nightscout?

paul1956, 2023-08-19:
@bewest I don't understand some of your comments. The defensive warnings were copied, I am not a lawyer. If you have specific changes, I would be happy to accept. I would he thrilled if you add this to FDA remote monitor registry.

As for sync with NightScout I have no idea what that entails I have never used NightScout personally. This code is very tailored to Medtronic 770G and 780G to allow a deeper understanding of the algorithm and information that pump makes available some of which is very hard or impossible to get from existing reports. 

I personally don't promote this app; I share it freely with a few users and they share it organically.  My next priority (which I am having 0 success with) is to read the CareLink DEVICE SETTINGS SNAPSHOT report to allow the program to automatically import that information which is not available from pump (today its manual entered) and additionally reformat it to be a step-by-step personalized report to program a replacement pump. It will merge the data from device setting with format of "How to program a new pump". This MAY cause liability to me and if so, I may just use this new report for myself as I get 4-6 new pumps a year due to travel schedule and am constantly missing a setting.

bewest, 2023-08-20:
Very reasonable.  Feel free to get in touch with me, bewest@medicaldatanetworks.com.  There is an additional project to connect to the BLE device directly, it sounds like you might be an ideal contributor; it could lead to restoring/configuring settings on the pump.  Finally, I'm not yet a dotnet dev, but I'm very impressed with MAUI and f#, and it would be great to ask you some questions about feasibility of open source and cross platform mobile apps in this space.

paul1956, 2023-08-21:
@bewest replied via email

closed 2023-09-05 by paul1956