paul1956/CareLink #47: first time user. not able to login with account of Australia.

kk001kk, 2023-08-29:


paul1956, 2023-09-05:
@kk001kk I need more information? Is there an error? Are there any files in MyDocuments/CareLink
I am also posting a new version today that might fix your issue.

Also, can you share the CareLink server URL. I assume it is http://carelink.minimed.eu

paul1956, 2023-09-05:
@kk001kk You can try this prerelease 
[CareLink3.9.0.1x64.zip](https://github.com/paul1956/CareLink/files/12518484/CareLink3.9.0.1x64.zip)

paul1956, 2023-09-10:
@kk001kk any update or I will close I have just posted a newer version, if it fixes your issue please close otherwise I need more information.

paul1956, 2023-09-23:
@kk001kk since I have no actionable information I will close. If you are still having issues please reopen.

closed 2023-09-23 by paul1956