paul1956/CareLink #49: Widget Bg level not displaying correctly

Ryptahi, 2023-10-02:
Seems like I have an issue that doesn't display the blood glucose level on widget. It does however display  the recent change to it, and also I can see the blood glucose level correctly when switching to main display.

[image]
[image]

paul1956, 2023-10-03:
@Ryptahi I believe this is fixed in the current beta, if not let me know.

This could also be related to incorrect data from CareLink after iOS 17 upgrade.  To verify if that is the issue you can save a Snapshot and post here all personally identifiable data is removed from the text file before saving.

Ryptahi, 2023-10-03:
> @Ryptahi I believe this is fixed in the current beta, if not let me know.
>
> This could also be related to incorrect data from CareLink after iOS 17 upgrade. To verify if that is the issue you can [...]
I encountered this issue in 3.9.2.5 Beta 2 version

Here is the snapshot
[CareLinkSnapshot(en-FI).txt](https://github.com/paul1956/CareLink/files/12788831/CareLinkSnapshot.en-FI.txt)

paul1956, 2023-10-04:
@Ryptahi thanks will review today.

paul1956, 2023-10-04:
@Ryptahi was it really 12:01? I am wondering if this is the CareLink bad time bug that is also effecting MiniMed Mobile and CareLink Connect. I will do more research today in any case.

paul1956, 2023-10-04:
I think I found it, please try CareLink3.9.2.5x64Beta3. If not, I will need a follower login if you want to share it privately.

Ryptahi, 2023-10-04:
> I think I found it, please try CareLink3.9.2.5x64Beta3. If not, I will need a follower login if you want to share it privately.

Yes, issue doesn't seem to be present in beta 3 release and everything seems to work correctly once again. Thank you so much once again :)

closed 2023-10-04 by paul1956