paul1956/CareLink #52: Occasional error when logging in

Ryptahi, 2023-11-22:
This error seems to be happen a bit randomly. I can login and everything works fine, but after few captcha solves this usually comes up and takes many hours before I can log back in again.  

Below is error log

--- Start of Exception ---
Value does not fall within the expected range.
--- End of Exception ---
--- Start of stack trace ---
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Page.CreateIsolatedWorldAsync(String frameId, String worldName, Nullable`1 grantUniveralAccess)
   at A.C.A(String)
   at WebView2.DevTools.Dom.WebView2DevToolsContext.A()
   at WebView2.DevTools.Dom.CoreWebView2Extensions.CreateDevToolsContextAsync(CoreWebView2 coreWebView2, ILoggerFactory loggerFactory)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NGXXXXXH"  // Its 780G pump
}

[image]

paul1956, 2023-11-26:
@Ryptahi  Fixed in CareLink4.0.0.5x64

paul1956, 2023-11-26:
@Ryptahi Fixed in CareLink4.0.0.6x64

Ryptahi, 2023-11-26:
> @Ryptahi Fixed in CareLink4.0.0.6x64

Hmm didn't get this error anymore, but now I faced another error.

[image]

--- Start of Exception ---
Class not registered

--- End of Exception ---
--- Start of stack trace ---
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Runtime.EvaluateAsync(String expression, String objectGroup, Nullable`1 includeCommandLineAPI, Nullable`1 silent, Nullable`1 contextId, Nullable`1 returnByValue, Nullable`1 generatePreview, Nullable`1 userGesture, Nullable`1 awaitPromise, Nullable`1 throwOnSideEffect, Nullable`1 timeout, Nullable`1 disableBreaks, Nullable`1 replMode, Nullable`1 allowUnsafeEvalBlockedByCSP, String uniqueContextId)
   at WebView2.DevTools.Dom.ExecutionContext.A(Boolean, String)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
"medicalDeviceSerialNumber": "NGXXXXXH" // Its 780G pump
}

paul1956, 2023-11-26:
@Ryptahi 
Does it say .6 in version? Do you have .Net 8.0 installed?
[image]

Ryptahi, 2023-11-26:
@paul1956 Yes. It also prompted me to install the .Net 8.0 which I did. The error happens every time I log in right now

paul1956, 2023-11-26:
@Ryptahi Try updating Webview2 https://developer.microsoft.com/en-us/microsoft-edge/webview2/consumer/

paul1956, 2023-11-26:
@Ryptahi I hope all this ReCapchta nonsense will be gone by the end of the year. Sorry for your issues.

Ryptahi, 2023-11-26:
@paul1956 I will try to update it Thanks. Yeah i truly hope it will be over soon, but knowing medtronic they probably will just keep it....

paul1956, 2023-11-26:
@Ryptahi not if caregivers complain continue to complain loudly as they have been.

paul1956, 2023-12-10:
I believe this is fixed and I am closing, if not please reopen

closed 2023-12-10 by paul1956