paul1956/CareLink #53: i cannot log in....

SofiaKar1, 2023-12-10:
--- Start of Exception ---
Η κλάση δεν έχει καταχωρηθεί

--- End of Exception ---
--- Start of stack trace ---
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Runtime.EvaluateAsync(String expression, String objectGroup, Nullable`1 includeCommandLineAPI, Nullable`1 silent, Nullable`1 contextId, Nullable`1 returnByValue, Nullable`1 generatePreview, Nullable`1 userGesture, Nullable`1 awaitPromise, Nullable`1 throwOnSideEffect, Nullable`1 timeout, Nullable`1 disableBreaks, Nullable`1 replMode, Nullable`1 allowUnsafeEvalBlockedByCSP, String uniqueContextId)
   at WebView2.DevTools.Dom.ExecutionContext.A(Boolean, String)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}

paul1956, 2023-12-10:
@SofiaKar1 this should be fixed in 4.0.0.6 combined with update Medtronic did. Close if its working.

SofiaKar1, 2023-12-11:
--- Start of Exception ---
Η κλάση δεν έχει καταχωρηθεί

--- End of Exception ---
--- Start of stack trace ---
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Runtime.EvaluateAsync(String expression, String objectGroup, Nullable`1 includeCommandLineAPI, Nullable`1 silent, Nullable`1 contextId, Nullable`1 returnByValue, Nullable`1 generatePreview, Nullable`1 userGesture, Nullable`1 awaitPromise, Nullable`1 throwOnSideEffect, Nullable`1 timeout, Nullable`1 disableBreaks, Nullable`1 replMode, Nullable`1 allowUnsafeEvalBlockedByCSP, String uniqueContextId)
   at WebView2.DevTools.Dom.ExecutionContext.A(Boolean, String)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}

IT IS NOT WORKING AGAIN

SofiaKar1, 2023-12-12:
i still cannit log in
i downloaded the latest version but still the ploblem remains
the message is the following
what can i do ???

 Start of Exception ---
Η κλάση δεν έχει καταχωρηθεί

--- End of Exception ---
--- Start of stack trace ---
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Runtime.EvaluateAsync(String expression, String objectGroup, Nullable`1 includeCommandLineAPI, Nullable`1 silent, Nullable`1 contextId, Nullable`1 returnByValue, Nullable`1 generatePreview, Nullable`1 userGesture, Nullable`1 awaitPromise, Nullable`1 throwOnSideEffect, Nullable`1 timeout, Nullable`1 disableBreaks, Nullable`1 replMode, Nullable`1 allowUnsafeEvalBlockedByCSP, String uniqueContextId)
   at WebView2.DevTools.Dom.ExecutionContext.A(Boolean, String)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}

nfontan, 2023-12-12:
@paul1956 Same here.
[CareLinkErrorReport(en-US)4.txt](https://github.com/paul1956/CareLink/files/13653241/CareLinkErrorReport.en-US.4.txt)
[CareLinkErrorReport(en-US)3.txt](https://github.com/paul1956/CareLink/files/13653242/CareLinkErrorReport.en-US.3.txt)
[CareLinkErrorReport(en-US)2.txt](https://github.com/paul1956/CareLink/files/13653243/CareLinkErrorReport.en-US.2.txt)
[CareLinkErrorReport(en-US)1.txt](https://github.com/paul1956/CareLink/files/13653244/CareLinkErrorReport.en-US.1.txt)

And some graphics bugs
[image]

SofiaKar1, 2023-12-14:
i cannot understand what i sould do in order to log in..
could you connect with anydesk to my laptop in order to help me ??

paul1956, 2023-12-15:
@SofiaKar1 I am at sea with very slow connection for 7 more days, after that I can help you. You might try Terry L Witt on Facebook he may be able to help while I am out.  Part of the issue is Medtronic keeps changing things to prevent 3rd party software, but I was working for me today.

paul1956, 2023-12-15:
@SofiaKar1, @nfontan given I have not changed anything since October 12th and this issue just started happening, I suspect Medtronic is again trying to block 3rd party applications or their servers are having issues that the current code does not handle it as well as previous versions. Someone else is looking at it until I return on 12/22. I have tried it several time and it works occasionally with no changes on my part.

paul1956, 2024-01-02:
@nfontan @SofiaKar1 Install 4.0.0.7 and see note, you need to wait for the puzzle and not chick the "I am not a robot box".

Please close if fixed.

nfontan, 2024-01-03:
> @nfontan @SofiaKar1 Install 4.0.0.7 and see note, you need to wait for the puzzle and not chick the "I am not a robot box".
> 
> Please close if fixed.

Hi Paul,
Do you remove the binary files?

I only see the source code.
Thanks

paul1956, 2024-01-03:
@nfontan Yes, I messed up, its fixed now. Sorry.

nfontan, 2024-01-03:
> @nfontan Yes, I messed up, its fixed now. Sorry.

I was able to download it, but I continue to have different errors

Thanks!

[image]
[image]
[CareLinkErrorReport(en-US)3.txt](https://github.com/paul1956/CareLink/files/13823948/CareLinkErrorReport.en-US.3.txt)
[CareLinkErrorReport(en-US).txt](https://github.com/paul1956/CareLink/files/13823949/CareLinkErrorReport.en-US.txt)
[CareLinkErrorReport(en-US)1.txt](https://github.com/paul1956/CareLink/files/13823950/CareLinkErrorReport.en-US.1.txt)
[CareLinkErrorReport(en-US)2.txt](https://github.com/paul1956/CareLink/files/13823951/CareLinkErrorReport.en-US.2.txt)

paul1956, 2024-01-03:
@nfontan I see that as well. Medtronic is blocking something that I can't get around until they open the API.

paul1956, 2024-01-09:
@nfontan I have found if you let the program fill in all the login information and automatically click "I am not a robot" it will work. If you have to fill in any information or click the box manually it will fail. It looks like Medtronic is sending 2 or more different reCAPTCHA HTML pages and my code only understands 1.  It everything is as expected the program will check the box and, in many cases, skip the reCAPTCHA. When it gets an unexpected page it just sites waiting.

IoanaSara, 2024-01-23:
I can connect to carelink with version 4.0.04.
I can't do it with higher versions.

paul1956, 2024-02-24:
@nfontan, @IoanaSara, @SofiaKar1, @DualBrain I believe version 4.0.0.11 Fixes the Captcha issue.

Please close if it does.

IoanaSara, 2024-02-24:
I tried logging in, but I got the attached error. Thank you very much!

    Pe sâmbătă,

 
 

  
--- Start of Exception ---
Class not registered

paul1956, 2024-02-24:
I will try to look at it this weekend. What language is Windows set to?Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

paul1956, 2024-02-25:
Try downloading and installing WebView2 directly from Microsoft (don’t trust random links on GitHub) it should not be necessary but it may fix this issue. Microsoft Edge WebView2 | Microsoft Edge Developerdeveloper.microsoft.comSent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

IoanaSara, 2024-02-25:
I greet you with respect, - Windows 11 operating system - WebView2 is installed- Romanian windows language- I changed the language to English and send you the file with the error.I manage to connect with CareLink version 4.0.0.5x64Thank you very much!

    Pe duminică,

nfontan, 2024-02-27:
I have the same error from webview2. I also have windows 11 but in Spanish

[image]

[CareLinkErrorReport(en-US)2.txt](https://github.com/paul1956/CareLink/files/14412971/CareLinkErrorReport.en-US.2.txt)

paul1956, 2024-02-27:
Several people are having issues on non US servers, don’t think it’s language related. I hope to look at it today or tomorrow. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

image.png (view on web)

paul1956, 2024-02-27:
There is something changed in the Flow outside North America after successful login, it is redirected to log in again.

@nfontan I will need some help figuring this out. This issue for anyone that wants to try is in LoginDialog.vb line 379. it should only be called once but is called a second time and cause a crash.

paul1956, 2024-02-27:
@nfontan I have improved some of the debug output, it I run latest version (4.0.0.14) 10 times on EU server 8 fail and 2 work with me changing nothing. On US all 10 work.

IoanaSara, 2024-03-02:
Hello,I tried version 14, no error, thank you very much!I am interested in seeing the Safe Meal Reduction.In order to see it, I have to proceed as follows:1. Start Here -> Use Test Data;2. Start Here -> Use Last Data File;After this artifice, the Safe Meal Reduction value is displayed!Thank you so much!

    Pe marți,

nfontan, 2024-03-02:
> @nfontan I have improved some of the debug output, it I run latest version (4.0.0.14) 10 times on EU server 8 fail and 2 work with me changing nothing. On US all 10 work.

Thanks Paul
On version 14 login works good! Congrats!

one questian, How do I change the duration of active insulin now?
[image]
The box is disable, Thanks!

paul1956, 2024-03-03:
I would like to take credit but I think it continues to be server issues. Different servers are sending different replies, and if you get an odd reply I don’t recover. At least it doesn’t crash on bad server data.Version 14 allows cleanup of all the junk left over from previous crashes under “Start Here” menu.The active insulin should be loaded automatically from your last CareLink settings report (English only). There is a PDF file under MyDocuments/Carelink that I download from CareLink and import all the data, it also allows creation of a personal pump setting report. It should update that file when it changes on server. If you delete that file you can download one manually from CareLink with web browser (to download directory) and then using menu import it (it moves file and renames it).If you want to fake a different number that might be disabled since it now gets it directly from pump.Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

image.png (view on web)

paul1956, 2024-03-03:
Replied to another message with details check GitHub, I can’t access GitHub till 3/23 this see your image.png. Ship broke GitHub Access, working with IT to fix. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

paul1956, 2024-04-03:
Fixes

closed 2024-04-03 by paul1956