paul1956/CareLink #54: non mi fuziona carelik su windows

liberogiacomelli2017, 2024-01-02:
buongiorno paolo l errore che mi da e questo        --- Start of Exception ---
Interfaccia non registrata.

--- End of Exception ---
--- Start of stack trace ---
   at System.Runtime.InteropServices.Marshal.ThrowExceptionForHR(Int32 errorCode)
   at Microsoft.Web.WebView2.Core.CoreWebView2.CallDevToolsProtocolMethodAsync(String methodName, String parametersAsJson)
   at Microsoft.Web.WebView2.Core.DevToolsProtocolExtension.Runtime.EvaluateAsync(String expression, String objectGroup, Nullable`1 includeCommandLineAPI, Nullable`1 silent, Nullable`1 contextId, Nullable`1 returnByValue, Nullable`1 generatePreview, Nullable`1 userGesture, Nullable`1 awaitPromise, Nullable`1 throwOnSideEffect, Nullable`1 timeout, Nullable`1 disableBreaks, Nullable`1 replMode, Nullable`1 allowUnsafeEvalBlockedByCSP, String uniqueContextId)
   at WebView2.DevTools.Dom.ExecutionContext.A(Boolean, String)
--- End of stack trace from previous location ---
{
  "firstName": "First",
  "lastName": "Last",
  "medicalDeviceSerialNumber": "NG1234567H"
}
se mi puoi inviare una guida per email a satanicmekkanic@gmail.com

paul1956, 2024-01-02:
@liberogiacomelli2017 Install 4.0.0.7 and see note, you need to wait for the puzzle and not chick the "I am not a robot box".

Please close if fixed.

paul1956, 2024-01-03:
@liberogiacomelli2017 Added binary

liberogiacomelli2017, 2024-01-03:
Grazie domani ci provo buonaserata

liberogiacomelli2017, 2024-01-04:
Buongiorno paolo mi poi mandare il link della versione 4.0.0.7 grazie matteo

liberogiacomelli2017, 2024-01-04:
Paolo ho fatto istallato 4.0.0.7 ma il log in e la casella re capta me la compila da solo come posso fare??

liberogiacomelli2017, 2024-01-04:
[image]

paul1956, 2024-01-04:
@liberogiacomelli2017 After you fill out the login screen, the captcha form should popup with your username and password and then the "I am not a robot" check box should automatically click, and the puzzle should appear. If that does not happen there is nothing, I can do the issue is on the Medtronic side. If you click the button, it will fail.

liberogiacomelli2017, 2024-01-05:
Buongiorno e grazie prima cosa ho messo username e password poi automaticamente barra la casella non sono un robot risolvo il puzzle ma niente ni da la schermata che ti ho già inviato per adesso grazie riprovo molto gentile e auguri di buon anno matteo Giacomelli livorno italy

marcindanek85, 2024-02-19:
It seems that something is changed in login process.
It seems it enough to add try...catch to looking for textbox'es on login page:

`Dim userNameInputElement As HtmlInputElement = Nothing
Dim passwordInputElement As HtmlInputElement = Nothing
Dim loginButtonElement As HtmlInputElement = Nothing
Try
    userNameInputElement = Await Me.DevContext.QuerySelectorAsync(Of HtmlInputElement)("#username")
    passwordInputElement = Await Me.DevContext.QuerySelectorAsync(Of HtmlInputElement)("#password")
    loginButtonElement = Await Me.DevContext.QuerySelectorAsync(Of HtmlInputElement)("[name=""actionButton""]")
    If userNameInputElement Is Nothing OrElse passwordInputElement Is Nothing OrElse loginButtonElement Is Nothing Then
        Exit Sub
    End If
Catch
    Exit Sub
End Try`

marcindanek85, 2024-02-20:
Have You tried to remove the annoying reCaptcha when each login attempt?
I made to changes:
comment out this stuff:
`'' Clear cookies
'_httpClient.DefaultRequestHeaders.Clear()
'' Clear basic session records
'Me.SessionUser.Clear()
'Me.SessionProfile.Clear()
's_sessionCountrySettings.Clear()
'_sessionMonitorData.Clear()

' Open login(get SessionId And SessionData)`

and setting environment like (according to: [https://learn.microsoft.com/en-us/microsoft-edge/webview2/concepts/user-data-folder?tabs=dotnet#specifying-a-custom-udf-location](microsoft))
Dim _task As Task(Of CoreWebView2Environment) = CoreWebView2Environment.CreateAsync(Nothing,
                                    UserDataFolder,
                                    Nothing)

Await Me.WebView21.EnsureCoreWebView2Async(Await _task)

paul1956, 2024-02-20:
I am at sea until Friday with no access to GitHub. I will try Friday. You can also create a pull request (PR) that I can Merge. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

paul1956, 2024-02-22:
Change was a little different but your solution worked. I will update GitHub when we dock in Hawaii, on Saturday. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

paul1956, 2024-02-24:
@marcindanek85, @liberogiacomelli2017, @Krikke99 I believe version 4.0.0.11 Fixes the Captcha issue.

Please close if it does.

paul1956, 2024-02-28:
Version 14 seems to have helped but I think issue is different servers sending different responses. Paulina did find another issue which will cause a crash after a while and I can’t upload fix until I get to shore on 3/23/2024. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

paul1956, 2024-04-03:
Fixed

closed 2024-04-03 by paul1956