paul1956/CareLink #55: Unable to connect to carelink

IoanaSara, 2024-01-19:
[image]
[CareLinkErrorReport(en-US)95.txt](https://github.com/paul1956/CareLink/files/13992312/CareLinkErrorReport.en-US.95.txt)

paul1956, 2024-01-19:
It is a Medtronic server issue; they are blocking access by changing the code sent to client.
The only solution is to let the program fill out the Medtronic Login screen and click the "I am not a Robot".  It you fill out any information manually it will always fail. If it does not automatically fill it in cancel and try again usually within 2 tries it will work.

This is a duplicate of #53

closed 2024-01-19 by paul1956

IoanaSara, 2024-01-21:
I can only connect with version 4.0.0.4. With higher versions it doesn't work, the credentials are filled in automatically.