paul1956/CareLink #56: Data unavailable Safe Meal Reduction.

IoanaSara, 2024-03-10:
Data from the Safe Meal Reduction report is not available.

[image]

paul1956, 2024-03-10:
Save meal reduction is a calculation base on your CareLink Settings file. If it is unable to download an English settings file it will not display SafeMealReduction. You can download the file manually and then use the programs Start Here menu to import the file. It expects the file in your download directory. Safe meal reduction is a calculation based on the size of your bolus based on pump settings and what is delivered.

 

If you are seeing something different let me know.

 

Safe.Meal.Reduction.png (view on web) <https://github.com/paul1956/CareLink/assets/113257829/4b00cae0-a931-4c1e-bbdf-0bb4f550b4e9>

paul1956, 2024-04-03:
Fixed

closed 2024-04-03 by paul1956