paul1956/CareLink #57: Data difference at expoort in excel file

IoanaSara, 2024-03-10:
When exporting the data from Auto Basal Delivery Markers, I noticed that the Basal Rate values ​​in the excel file are not identical to those in Carelink.
[image]

paul1956, 2024-03-11:
Can you post a saved snapshot file that I can export to excel to see difference? Snapshots have all personal information removed. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

AutoBasalExcel.png (view on web)

IoanaSara, 2024-03-11:
Hello,I hope the attached files are good.I notice that the differences are 
Thank you very much!

    Pe luni,

paul1956, 2024-03-11:
@IoanaSara
Great catch the faction that was rounding Basal values was very messed up with certain values and .075 was rounding to .050 is just one example.
Version CareLink4.0.0.15x64 fixes it and should also fix Meal Reduction

Fix so please close one of both issues.

IoanaSara, 2024-03-11:
Hello,Version 4.0.0.16 solved the rounding problem, there are no more differences.Safe Meal Reduction data are not yet available.To view the Safe Meal Reduction, it is necessary to access: Start Here -> Use Test Data; Start Here -> Use Last Data File.Thank you very much!

    Pe luni,

paul1956, 2024-04-03:
Fixed