paul1956/CareLink #58: Session expires after 35 minutes

mattimart, 2024-04-03:
Carelink is working properly but after about 35 minutes the session expires and the program stops updating and needs a new login
[image]

paul1956, 2024-04-03:
@mattimart Fixed in CareLink4.0.1.1x64. Thanks for the catch.

Please close if you agree.

maxdego, 2024-04-04:
new version CareLink4.0.1.1x64 downloaded and installed, but the problem still remains

mattimart, 2024-04-04:
> @mattimart Fixed in CareLink4.0.1.1x64. Thanks for the catch.
> 
> Please close if you agree.

Carelink.exe in release 4.0.1.1 is still 4.0.1.0 
[image]

maxdego, 2024-04-04:
Carelink.exe release 4.0.1.1 is still 4.0.1.0
Yes, I had already noticed it as soon as I installed it, but I thought the login problem had been solved

paul1956, 2024-04-04:
Are you saying version is wrong or issue is still there or both?I will fix version number today to 4.0.1.2. And also verify issue 35 minute failure is fixed. The fix in 4.0.0.1 was to address new SmartGuard information in an updated pump in Ireland, it only effected users there.Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

maxdego, 2024-04-04:
both, version is wrong and issue is still there

paul1956, 2024-04-04:
The timeout of 35 minutes (it was 24 hours) is controlled by Medtronic, the same happens with a Browser but you should be able to redo Capchta and continue if not let me know.

closed 2024-04-05 by mattimart