paul1956/CareLink #59: Connection error after Medtronic maintenance.

IoanaSara, 2024-04-06:
After the maintenance performed by Medtronic, it is no longer possible to connect the application.
Carelink Europe.
[image]

IoanaSara, 2024-04-06:
I checked again, it's ok now.

paul1956, 2024-04-06:
@IoanaSara I put in a fix for that error to prevent crash, in latest version, but yes they caused it and fixed it. They were not sending any data but saying it was OK.

closed 2024-04-06 by paul1956

IoanaSara, 2024-04-07:
Hello,Thank you very much for everything you do for us.We still have problems with sensor calibration, most of the time they are unsupported calibrations. The ISIG value is also recorded in the Carelink report. Is it possible to display this ISIG value read from the sensor - online value? Or another value to help me calibrate the sensor.Thank you very much!

    Pe sâmbătă,

paul1956, 2024-04-07:
@IoanaSara what report is it on, you can look at all the raw date by opening documents/CareLink/CareLinkLastDownload(en-US).json where en-US depends on your language and country.

I think ISIG is only available on 770G, The 780G does not send it or use it. Do you have G4 or GL3? With G4 I you don't calibrate.

If you have a 770G you can use CalFactor.com to see if calibration will be accepted. You can also try marinating sensor up to 6 hours while wearing old sensor.

IoanaSara, 2024-04-08:
Thank you for your kindness!
In the csv report from CareLink, all SG values ​​are accompanied by ISIG Value. I have a lot of Sensor Calibration Rejected on both G3 and G4. I also calibrated the G4 because there are very big differences compared to the glucometer (difference>200 mg/dl). Many times it doesn't even accept my first calibration after heating the sensor.
I'm going to try the marinating sensor, now I've read about this procedure, I didn't know.

Thank you very much for everything!

    Pe luni,

paul1956, 2024-04-09:
@IoanaSara I see them in the CareLink report for some values in my report but not all, but they are not available in the stream I have access to. It's also interesting that they are all between 4-6, and CalFactor will fail see https://www.cal-factor.com with values outside 3-8.

All my values are 4.9-6.0 well within 3-8.