paul1956/CareLink #60: constant login requests with related CAPTCHA.

maxdego, 2024-04-13:
Hi everyone, I'm very sorry but I'm forced to abandon the use of the beautiful app due to the constant login requests with related CAPTCHA. I also note that if you don't want to log in, it's impossible to exit the app... until you fill in the CAPTCHA photos. Thanks anyway for the great work done
(for info "auto login" always selected)
ver. in use 4.0.1.2 x 64 but from file donwloaded result 4.0.1.3 x 64

paul1956, 2024-04-13:
@maxdego Yes Medtronic changed timeout from 24 hours to ~ 30 minutes and made the CAPTCHA much more difficult to solve. It is the same on CareLink Website and nothing I can do. I will be working on more analysis features that work on a snapshot over the coming months as well as hoping Medtronic opens API but without a lot of customers complaining I don’t see that happening.

closed 2024-04-13 by paul1956

maxdego, 2024-04-13:
no no, I'm not complaining to those who developed this wonderful app, I'm only complaining about the fact that it has become almost impossible to use it (but it's Medtronic's fault) I remain of the opinion that you are doing an excellent job
I hope for a resolution of the problem, because you can't give up such a useful app