paul1956/CareLink #61: Axis Object - The minimum value of the axis is greater than the largest data point value.

BennMcKenzie, 2024-06-24:
System Log's in then loads some text data and pump settign ssuch as AIT , Serial Number data but fails with above exception whilst trying to load the Main view and shows a big white box with a red X through it then crashes
[CareLinkErrorReport(en-NZ)4.txt](https://github.com/user-attachments/files/15949049/CareLinkErrorReport.en-NZ.4.txt)

BennMcKenzie, 2024-06-24:
If I don't login then change to another view I can Login and go through the other views its only the Main screen

paul1956, 2024-06-24:
Version 4.0.1.4 fixes the issue it will be posted soon. Please close if issue is fixed

BennMcKenzie, 2024-06-24:
Hi Paul, 

You are correct with time/date changes but your latest version has 100% fixed the issue

closed 2024-06-24 by BennMcKenzie