paul1956/CareLink #62: login unsuccessful

maxdego, 2025-02-08:
since yesterday the app refuses the login....

paul1956, 2025-02-08:
Yes, Medtronic changed the interface to block this and other applications 3rd party applications.
Working on it, this will continue to happen unless people complain to Medtronic that they want access to their data.

maxdego, 2025-02-09:
> @maxdego Yes, Medtronic changed the interface to block this and other

⁣tks Paul...
I'm waiting for new updates, it's impossible to do without this app now

regards
Max

paul1956, 2025-03-27:
@maxdego Version 5.0.0.0 works with new server code.
 I have only tested US Server but the code is there to handle all countries.

maxdego, 2025-03-28:
thanks 1000 Paul for the new version

 

I tried the login, (Europe-Italy server) but unfortunately it doesn't work

it keeps giving me a dialog box with the message

 

"ERROR: failed to refresh token, status 400"

do you want to try logginng in again?

 

I select yes, but the dialog box reappears 3-4 times without success, then the program closes definitively

 

I hope I was helpful, to solve the problem

I await news

 

regards

Max

paul1956, 2025-03-29:
I updated it 5.0.0.4 should fix that (when I post it) but you need to delete the json files in documents/carelink/yourUserUI.

maxdego, 2025-03-29:
thanks Paul,

as per your suggestion, I deleted the .json file and the login problem was solved.

but now there is another problem... the insulin data in the box of the last 24 hours are wrong.

 

for example:

carbs show =173 grams ok

carbs real =173 grams ok

basal- show =6885,0U (it is impossible)

meal bolus- show =2905,0U

meal bolus- real =29,05U

auto correct.- show =1038,0U

auto correct.- real =13,8U

total insulin -show =10828, U (it is impossible)

 

is it possible to solve?

I hope I was helpful

 

regards

Max

paul1956, 2025-03-29:
@maxdego It look like it has to do with your number formatting. Comma vs period as decimal separator. If you open the last downloaded json file (notepad works) from documents/CareLink replace any personal data like name, and anything else you aren't comfortable sharing publicly with any characters and numbers with random numbers. You can attach  file to issue.  Just don't change the length of the file. All the personal information is in the first few lines.

Also download latest CareLink setting file and use menu in program to import it. That will make results more accurate. Also make sure insulin type is correct.

paul1956, 2025-03-29:
@maxdego 5.0.0.6 will do parsing for comma decimal separator. I really need a Json file to test further, your file is documents/careling/CareLinkLastDownload(en-US).json

What you need to clean up is below

        cleanRecentData("firstName") = "First"
        cleanRecentData("lastName") = "Last"
        cleanRecentData("medicalDeviceSerialNumber") = "NG1234567H"
        cleanRecentData("conduitSerialNumber") = New Guid().ToString
        cleanRecentData("systemId") = "40000000000 0000"
        cleanRecentData("deviceSerialNumber") = "NG4000000H"

maxdego, 2025-03-30:
good morning Paul,
here is the file you requested
i hope everything went well

regards
Max

[CareLinkLastDownload(it-IT).json](https://github.com/user-attachments/files/19524469/CareLinkLastDownload.it-IT.json)

paul1956, 2025-03-30:
@maxdego thanks
Home page looks good now

[image]

I will update version tomorrow, but the version online should get you started.

paul1956, 2025-04-01:
@maxdego Lots of fixed some for your issues in version 5.0.0.10, if your log-in issues is fixed please close this issue.

closed 2025-04-01 by maxdego

paul1956, 2025-04-02:
@maxdego I can't reproduce anymore with the latest version. I am sure it's related to the use of comma as decimal separator (it-IT) VS (en-IT). If you save a snapshot and attach to NEW issue using latest version. Snapshots now filter out all personal information. I need to trace from raw stream to display.

maxdego, 2025-04-02:
hi Paul,
this afternoon I did some tests, I think I found the problem,
1) if I change the language of windows from Italian to United States, everything works correctly
or
2) if I do not change the language of windows, but I leave the Italian language, and I only change the decimal symbol from comma to dot, everything works correctly

can it be solved?
thanks again for your support

[image]

paul1956, 2025-04-02:
@maxdego Yes somewhere I am missing a place I need to translate comma to dot before I parse a string or and reverse when display a value. There an 100's of places where values are translated from string to numbers and back. I am missing some. I will have time to look starting Saturday.

If you have screen shots of places it's wrong please add to this issue.

paul1956, 2025-04-02:
@maxdego do you use mg/dL or mmol? I think mmol is also an issue but I don't have any way to test that. The US servers don't support it so I have no access to a data file.

paul1956, 2025-04-04:
@maxdego look for new version tonight. I think I addressed all your issues. It turns out the raw data uses period as decimal separator and mm/dl for everyone. The app is responsible for conversion and display. I was doing culture specific parsing of the raw data.

At this point all I have left error message translation and better time change support. Originally messages were English strings now they are numbers with no hint so to what they mean. I will add a popup so if someone gets a new message they can log an issue and I can fix. I will be doing  about 12 time changes in the next 42 days so I should get it fixed by time I return.

maxdego, 2025-04-12:
hi Paul
I just installed the new version 5.0.1.0 but during the login the problem occurs that every (NEW DAY) if I don't delete the file xxxxxxxLogindata.json it gives me an error 200
every day the same problem
[image]

if instead after deleting the file, I try to do one or many logins (on the same day) everything works correctly, but if the date changes the problem repeats itself, and I solve it only by deleting the file xxxxxxLogindata.json.
(the problem also occurred in previous versions, I think from ver.5.0.0.0)
is it possible to do an auto delete of the file?
for the rest everything seems to work well, even the problem of the comma and the dot is now solved

many thanks for all

paul1956, 2025-04-12:
I could auto delete every day or only when I get error 200 but it should not be necessary the only time I have seen that is with people running Guardian Monitor the 2 programs conflict if both using patient ID.  I will look into it I have a lot of other fixes going into a release tomorrow. Sent from my iPhoneI apologize for any typos Siri might have made.(503) 803-6077

Immagine.2025-04-14.072302.jpg (view on web)

Immagine.2025-04-14.072302.jpg (view on web)

paul1956, 2025-04-12:
@maxdego if you don't mind I will add a dialog ID before fixing the issue unless I can figure out where this is happening. I don't see it and the same dialog happens in multiple places.

The next version fixes a lot of color contrast issues and hides some useless information.

paul1956, 2025-04-12:
Status code 200 is success so the error makes no sense. 

[image]