paul1956/CareLink #63: I can't connect to Carelink.

IoanaSara, 2025-05-14:
Hello,
I've been trying lately and I can't connect to the Carelink app. I'm using version 5.0.1.2.
The error returned when logging in is I've attached it in the two files.

[CareLinkErrorReport(en-US)saraenescu.txt](https://github.com/user-attachments/files/20197451/CareLinkErrorReport.en-US.saraenescu.txt)

[image]

IoanaSara, 2025-05-14:
> Try deleting the json file(s) that contains saraenescu under mydocuments/Carelink.
>
> I need to automate this, it happens when they upgrade servers.
>
> I will look into this more today to verify that is the only issue.

Hello,
I deleted the files and everything is ok now
Thank you very much!
Regards, Daniel!

    Pe miercuri,

paul1956, 2025-05-14:
@IoanaSara I will leave this open to automate the fix. Great that it's working.

paul1956, 2025-05-15:
@IoanaSara Thanks for finding this, I have a fix in version 5.0.1.5. If refresh of token fails required files will be deleted and you will be prompted to log in again.

closed 2025-05-15 by paul1956